OpenEVSE/ESP8266_WiFi_v2.x #154: use a minifier on the .js .html files

davehun, 2018-02-14:
the src/data contents files are minified in git so they are pretty hard to compare and debug (e.g. spotting changes and regressions).
They should be pretty in the source code and then minified before final upload. This would also make the final files much smaller

glynhudson, 2018-02-14:
`src/data` is generated from `src/html` on platformio compile. Any changes to `src/data` will be overwritten on next compile. Best to edit `src/html`. 

`src/data` is commited to git (not best practice I realise) to allow easier uploading to ESP

closed 2018-02-14 by glynhudson

jeremypoulter, 2018-02-14:
Arduino IDE support rather.

Kind of on this subject, I am not overly happy with how this works, it was
structured to not depend installation of additional tools, ie depend on
Python or online tools. Unfortunately these are not that good and don't
generate map files so it may be nice to move to something for mainstream,
eg https://webpack.js.org, adding a dependency on node.js.

Any thoughts?