OpenEVSE/ESP8266_WiFi_v2.x #156: Date sync bug still occurs but not as badly

ForumLeech, 2018-02-23:
Date continues to sync incorrectly when using wifi firmware v2.6.0 but it's not as bad as previous firmware. It consistently syncs to tomorrow when clicking the 'Sync' button. Problem persists after doing a restart on both modules.

[image]
1, 2018.
[image]

glynhudson, 2018-02-23:
Thanks for reporting. We will take a look. For reference the original issue: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/119 which should have been fixed by https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/132 by @joverbee.

Can you replicate this issues @joverbee ?

glynhudson, 2018-03-11:
Sorry, but I'm unable to replicate. Could you try the latest version V2.7.0 and confirm if this is still an issue: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.7.0

I'm using Chrome 64.0.3282.190 64bit. Also tested on Firefox and Android.

ForumLeech, 2018-03-12:
> ESP8266_WiFi_v2.x/releases/tag/2.7.0

Yes the date and time sync are working perfectly now after upgrading to
v2.7.0. Thanks for notifying me!

closed 2018-03-12 by glynhudson