OpenEVSE/ESP8266_WiFi_v2.x #160: Add nice iOS icon

sandeen, 2018-03-01:
I recently added the link to my OpenEVSE to my iphone home screen.  It'd be really nice to have a pretty icon for this.  :)

Apparently a handful of sizes is ideal, for various devices, for example (likely not the recommended sizes):

```
<link rel="apple-touch-icon" sizes="57x57" href="apple-icon-57x57.png" />
<link rel="apple-touch-icon" sizes="72x72" href="apple-icon-72x72.png" />
<link rel="apple-touch-icon" sizes="114x114" href="apple-icon-114x114.png" />
<link rel="apple-touch-icon" sizes="144x144" href="apple-icon-144x144.png" />
```

but if there's not room for that many, just adding:

`<link rel="apple-touch-icon" sizes="32x32" href="/favicon-32x32.png" />`

to use the existing favicon might be nice.

[Configuring Web Applications for iOS](https://developer.apple.com/library/content/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html)

[Currently preferred icon sizes](https://developer.apple.com/ios/human-interface-guidelines/icons-and-images/app-icon/)

If there's a higher resolution source for the current favicon I'll play with this if people think it's a good idea.
Thanks,
-Eric

glynhudson, 2018-03-01:
Sounds great. Any change of a pull request to ensure your exact working code is merged? I don't have an ios device to test on

sandeen, 2018-03-02:
I haven't done it yet but if you're game I will.  Is there a hi-res version of the logo somewhere?  Are there concerns about spiffs space restraints?  (actually I've lost the thread on where the html content is even stored these days ...)

glynhudson, 2018-03-07:
the html files are stored in `html/` then the minified js versions are generated on platform io compile and stored in `/data`. Any changes should be made to `html/`.

Yes, resources on the ESP are an issue. How large will the hi-res logo be? What is the pixel dimensions? 

@chris1howell do you have a copy of the openevse logo in high-res?

Suriv, 2019-01-08:
https://realfavicongenerator.net/

Twitter image:
[image]

Configuration:
Favicon for iOS - Web Clip
[image]

Favicon for Android Chrome
[image]

Windows 10
[image]

macOS Safari
[image]

HTml5
<link rel="apple-touch-icon" sizes="180x180" href="apple-touch-icon.png">
<link rel="icon" type="image/png" sizes="32x32" href="favicon-32x32.png">
<link rel="icon" type="image/png" sizes="16x16" href="favicon-16x16.png">
<link rel="manifest" href="site.webmanifest">
<link rel="mask-icon" href="safari-pinned-tab.svg" color="#5bbad5">
<meta name="msapplication-TileColor" content="#ffffff">
<meta name="theme-color" content="#ffffff">

packed
[favicon_package_v0.16.zip](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/2738690/favicon_package_v0.16.zip)

closed 2020-07-28 by glynhudson