OpenEVSE/ESP8266_WiFi_v2.x #164: Improve packaging web file

jeremypoulter, 2018-03-06:
I am not overly happy with how the minification works, it was structured to not depend installation of additional tools, ie depend on Python or online tools. Unfortunately these are not that good and don't
generate map files so it may be nice to move to something for mainstream, eg https://webpack.js.org, adding a dependency on node.js.

Any thoughts?

glynhudson, 2018-03-07:
I'm happy to move to a more mainstream solution. webpack looks like a popular and well maintained project with lots of github stars. I'm not familiar with it myself. What's the main issue with the current implementation? What do the map files to? 

Do you think we should continue to commit the generated files to github? They regularly cause merge conflicts. What this to maintain Arduino IDE comparability? 

What are these auto-generated files doing? Do these need to be committed?: 

```
src/web_server.style.css.h
src/web_server.wifi_portal.js.h
src/web_server.wifi_portal.htm.h
```

jeremypoulter, 2018-03-07:
> What's the main issue with the current implementation? 

Mainly the lack of the map files, but is also not very elegant. 

> What do the map files to?

The map files tell the debugger which unminified source files map to the minified source the browser is seeing, very handy for debugging the frontend.

> Do you think we should continue to commit the generated files to github?

Ideally not it is bad practice to commit auto generated files.

> What this to maintain Arduino IDE comparability?

Yes that is the main reason, also if we did go down this route it would remove the Node.JS dependency for those that just want to build or change the C/C++ code

> What are these auto-generated files doing?

They are the 'C' version of the files, just makes it easy to include in the binary.

glynhudson, 2018-03-07:
>The map files tell the debugger which unminified source files map to the minified source the browser is seeing, very handy for debugging the frontend.

Ah yeah, this is an issue with debugging. I would be really great to be able to link to the unminified source while debugging. I didn't realise this was an option, I assumed making it harder to debug was a side effect of the minification process. 

> Ideally not it is bad practice to commit auto generated files.

I'm happy to drop the auto generated files from the repo. It will make the repo less confusing to contributors and make merging easier. Do you have any thoughts on this @chris1howell? Compiling via platformio will be a requirement to build the project. 

> Yes that is the main reason, also if we did go down this route it would remove the Node.JS dependency for those that just want to build or change the C/C++ code

True, however I can't think of many changes a users would want to do that did not involve editing the js/html. 

>They are the 'C' version of the files, just makes it easy to include in the binary.

Is it also possible to drop them from the repo? i.e add to gitignore?

glynhudson, 2020-07-28:
@jeremypoulter can this be closed?

jeremypoulter, 2020-07-29:
Yeah this is done

closed 2020-07-29 by jeremypoulter