OpenEVSE/ESP8266_WiFi_v2.x #165: Simulator error

glynhudson, 2018-03-07:
I've just updated the simulator on our sever to the latest version. It seems to throws a js error:

`Uncaught ReferenceError: Unable to process binding "text: function (){return itemsLoaded }"
Message: itemsLoaded is not defined` 

https://openevse.openenergymonitor.org/

[image]

Interestingly when I run the simulator locally it runs fine. The error must either be due to serving via apache or https. Have you got any ideas?

jeremypoulter, 2018-03-07:
Sounds like the HTML resources are out of date.

jeremypoulter, 2018-03-07:
Actually, is working fine for me...

glynhudson, 2018-03-07:
Ah, also seems to be working fine for me now! Could have been a cloudflare for browser cache issue.

closed 2018-03-07 by glynhudson