OpenEVSE/ESP8266_WiFi_v2.x #167: Date off by one day because of timezone

moeabm, 2018-03-25:
Im in pacific (-8) timezone. Date field on web gui shows +1 day from 4p-midnight. Should localize time before filling in field. Tested on chrome and FF.

glynhudson, 2018-03-25:
What version are you running? 

Could this be a replicate of issue https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/156 that's been fixed in V2.7.0?

moeabm, 2018-03-25:
Im on 2.6.0
let me update and check back. Suspect you're right.

moeabm, 2018-03-25:
Looks like I still have the issue. 

[image]

glynhudson, 2018-03-26:
Oh dear, what time was your browser showing when you clicked 'sync' on the screen gab above? 06:16? Does a manual time set work for you? What browser are you using? Assuming you have tested on a couple of different browsers?

moeabm, 2018-03-26:
The time field is accurate. It's the date that is always in GMT. At the time of this screen grab, local time was 3/24/2018 @7:16pm. The system time is correct using auto sync (I've checked the front panel) but the wifi gui is reporting GMT date. Im using chrome but I've also tested firefox with the same result.

moeabm, 2018-03-26:
For example, currently, it's 7:15a local time on 3/26. Since GMT is also 3/26, my unit looks accurate. Come 5pm local time, it will change to 3/27/2018.

ForumLeech, 2018-04-09:
Maybe it would be easier and better for ESP8266 to use a time server like time.google.com. A time server's output is simple and stable and knows the daylight savings time schedule for every time zone. 
Host web browsers and OS's, as a time source, create challenges as they are multi-factorial and changing.

chris1howell, 2018-04-09:
This may be a really good idea. If the ESP8266 was keeping time and running
timers there would be no need for the RTC chip. Eliminating the RTC, timers
and menus would free up a ton of resources on the OpenEVSE controller.

jeremypoulter, 2018-04-09:
This issue would still happen irrespective of if the time came from NTP or OpenEVSE ;-) That being said Shifting the timers etc to the ESP8266 with NTP may have advantages probably should raise as a separate issue about this. IMHO I would only do this if the ESP was less of an optional extra, ie maybe do  new LCD board with the ESP module on it.

glynhudson, 2018-04-09:
@moeabm this issues should be now fixed in V2.7.1. Please could you test. Thank you 

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.7.1

ForumLeech, 2018-04-17:
Previous issue fixed by 2.7.0 resurfaced in 2.7.1. It syncs to a date one month in the past. It was 1:45 AM Pacific Time on 4/17/2018 when I tested it. Happens in multiple web browsers and operating systems. Restart didn't help. Can anybody replicate this bug resurfacing?

glynhudson, 2018-04-17:
@ForumLeech thanks for reporting. Yes, your totally right. I'm able to re-create this. 

@jeremypoulter are you able to re-create the month syncing one month in the past in 2.7.1? Would you like me to open a new issue for this?

closed 2018-05-01 by glynhudson