OpenEVSE/ESP8266_WiFi_v2.x #168: Delay timer field too small when

moeabm, 2018-03-25:
Start/Stop Timer fields cut off the AM/PM text con chrome due to 12-hour clock control. Can we make these a bit wider to accommodate chrome's up/down controls?  Firefox shows 24-hour clock so its a non-issue.

moeabm, 2018-03-25:
https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/time

glynhudson, 2018-03-25:
Sure, I guess it would not be a difficult change. However when I test with Chrome and Android they both show 24hr clock so AM/PM not an issue. My preferance would be to always use 24hrs clock  

[image]

moeabm, 2018-03-25:
I also prefer 24-hour clock but it seems to be browser defined.

PR #169 (2018-03-26): chaged time field width
show am/pm in some browsers
Fixes #168
closed 2018-03-26 by glynhudson