OpenEVSE/ESP8266_WiFi_v2.x #175: Octapus Energy Integration

glynhudson, 2018-04-17:
[Octopus Energy](https://octopus.energy) have recently launched an electricity tariff in the UK with dynamic pricing [Octapus Agile Tariff](https://octopus.energy/agile/) which included 'Plunge' pricing which occurs when energy prices go negative when there is too much power! At these times a user would get paid to use electricity. 

Octopus Energy have a developer API: https://developer.octopus.energy/docs/api/

It would be great to be able to integrate with the API to set the OpenEVSE to charge at the cheapest time of day and start charging when a 'Plunge' event occurs.

beaylott, 2018-04-17:
The relevant API call is:
https://api.octopus.energy/v1/products/{product_code}/electricity-tariffs/E-1R-AGILE-18-02-21-{gsp_id}/standard-unit-rates/

Where product_code for the Agile tariff is AGILE-18-02-21 (I dont think this will change but worth bearing in mind it is a parameter) and where gsp_id corresponds to the GSP Group ID as found in the MPAN (
https://en.wikipedia.org/wiki/Meter_Point_Administration_Number#Metered_Supply_Point). Note that some DNO's are associated with more than one GSP group.

So for a customer in GSP Group D (Merseyside and Northern Wales) the URL would be:

https://api.octopus.energy/v1/products/AGILE-18-02-21/electricity-tariffs/E-1R-AGILE-18-02-21-D/standard-unit-rates/

This should return a JSON like:

`{"count":22651,"next":"https://api.octopus.energy/v1/products/AGILE-18-02-21/electricity-tariffs/E-1R-AGILE-18-02-21-D/standard-unit-rates/?page=2","previous":null,"results":[{"value_exc_vat":9.24,"value_inc_vat":9.702,"valid_from":"2018-04-17T21:30:00Z","valid_to":"2018-04-17T22:00:00Z"},{"value_exc_vat":9.88,"value_inc_vat":10.374,"valid_from":"2018-04-17T21:00:00Z","valid_to":"2018-04-17T21:30:00Z"},{"value_exc_vat":11.53,"value_inc_vat":12.1065,"valid_from":"2018-04-17T20:30:00Z","valid_to":"2018-04-17T21:00:00Z"},{"value_exc_vat":14.61,"value_inc_vat":15.3405,"valid_from":"2018-04-17T20:00:00Z","valid_to":"2018-04-17T20:30:00Z"},{"value_exc_vat":15.18,"value_inc_vat":15.939,"valid_from":"2018-04-17T19:30:00Z","valid_to":"2018-04-17T20:00:00Z"},{"value_exc_vat":17.34,"value_inc_vat":18.207,"valid_from":"2018-04-17T19:00:00Z","valid_to":"2018-04-17T19:30:00Z"},{"value_exc_vat":14.96,"value_inc_vat":15.708,"valid_from":"2018-04-17T18:30:00Z","valid_to":"2018-04-17T19:00:00Z"},{"value_exc_vat":13.2,"value_inc_vat":13.86,"valid_from":"2018-04-17T18:00:00Z","valid_to":"2018-04-17T18:30:00Z"},{"value_exc_vat":28.4,"value_inc_vat":29.82,"valid_from":"2018-04-17T17:30:00Z","valid_to":"2018-04-17T18:00:00Z"},{"value_exc_vat":27.94,"value_inc_vat":29.337,"valid_from":"2018-04-17T17:00:00Z","valid_to":"2018-04-17T17:30:00Z"},{"value_exc_vat":26.2,"value_inc_vat":27.51,"valid_from":"2018-04-17T16:30:00Z","valid_to":"2018-04-17T17:00:00Z"},{"value_exc_vat":24.55,"value_inc_vat":25.7775,"valid_from":"2018-04-17T16:00:00Z","valid_to":"2018-04-17T16:30:00Z"},{"value_exc_vat":23.67,"value_inc_vat":24.8535,"valid_from":"2018-04-17T15:30:00Z","valid_to":"2018-04-17T16:00:00Z"},{"value_exc_vat":22.37,"value_inc_vat":23.4885,"valid_from":"2018-04-17T15:00:00Z","valid_to":"2018-04-17T15:30:00Z"},{"value_exc_vat":9.57,"value_inc_vat":10.0485,"valid_from":"2018-04-17T14:30:00Z","valid_to":"2018-04-17T15:00:00Z"},{"value_exc_vat":9.15,"value_inc_vat":9.6075,"valid_from":"2018-04-17T14:00:00Z","valid_to":"2018-04-17T14:30:00Z"},{"value_exc_vat":9.5,"value_inc_vat":9.975,"valid_from":"2018-04-17T13:30:00Z","valid_to":"2018-04-17T14:00:00Z"},{"value_exc_vat":9.57,"value_inc_vat":10.0485,"valid_from":"2018-04-17T13:00:00Z","valid_to":"2018-04-17T13:30:00Z"},{"value_exc_vat":9.39,"value_inc_vat":9.8595,"valid_from":"2018-04-17T12:30:00Z","valid_to":"2018-04-17T13:00:00Z"},{"value_exc_vat":9.57,"value_inc_vat":10.0485,"valid_from":"2018-04-17T12:00:00Z","valid_to":"2018-04-17T12:30:00Z"},{"value_exc_vat":9.46,"value_inc_vat":9.933,"valid_from":"2018-04-17T11:30:00Z","valid_to":"2018-04-17T12:00:00Z"},{"value_exc_vat":9.46,"value_inc_vat":9.933,"valid_from":"2018-04-17T11:00:00Z","valid_to":"2018-04-17T11:30:00Z"},{"value_exc_vat":9.24,"value_inc_vat":9.702,"valid_from":"2018-04-17T10:30:00Z","valid_to":"2018-04-17T11:00:00Z"},{"value_exc_vat":10.12,"value_inc_vat":10.626,"valid_from":"2018-04-17T10:00:00Z","valid_to":"2018-04-17T10:30:00Z"},{"value_exc_vat":9.66,"value_inc_vat":10.143,"valid_from":"2018-04-17T09:30:00Z","valid_to":"2018-04-17T10:00:00Z"},{"value_exc_vat":10.69,"value_inc_vat":11.2245,"valid_from":"2018-04-17T09:00:00Z","valid_to":"2018-04-17T09:30:00Z"},{"value_exc_vat":10.58,"value_inc_vat":11.109,"valid_from":"2018-04-17T08:30:00Z","valid_to":"2018-04-17T09:00:00Z"},{"value_exc_vat":11.68,"value_inc_vat":12.264,"valid_from":"2018-04-17T08:00:00Z","valid_to":"2018-04-17T08:30:00Z"},{"value_exc_vat":12.1,"value_inc_vat":12.705,"valid_from":"2018-04-17T07:30:00Z","valid_to":"2018-04-17T08:00:00Z"},{"value_exc_vat":13.64,"value_inc_vat":14.322,"valid_from":"2018-04-17T07:00:00Z","valid_to":"2018-04-17T07:30:00Z"},{"value_exc_vat":13.2,"value_inc_vat":13.86,"valid_from":"2018-04-17T06:30:00Z","valid_to":"2018-04-17T07:00:00Z"},{"value_exc_vat":9.88,"value_inc_vat":10.374,"valid_from":"2018-04-17T06:00:00Z","valid_to":"2018-04-17T06:30:00Z"},{"value_exc_vat":11.88,"value_inc_vat":12.474,"valid_from":"2018-04-17T05:30:00Z","valid_to":"2018-04-17T06:00:00Z"},{"value_exc_vat":11.44,"value_inc_vat":12.012,"valid_from":"2018-04-17T05:00:00Z","valid_to":"2018-04-17T05:30:00Z"},{"value_exc_vat":10.23,"value_inc_vat":10.7415,"valid_from":"2018-04-17T04:30:00Z","valid_to":"2018-04-17T05:00:00Z"},{"value_exc_vat":8.87,"value_inc_vat":9.3135,"valid_from":"2018-04-17T04:00:00Z","valid_to":"2018-04-17T04:30:00Z"},{"value_exc_vat":6.16,"value_inc_vat":6.468,"valid_from":"2018-04-17T03:30:00Z","valid_to":"2018-04-17T04:00:00Z"},{"value_exc_vat":6.6,"value_inc_vat":6.93,"valid_from":"2018-04-17T03:00:00Z","valid_to":"2018-04-17T03:30:00Z"},{"value_exc_vat":5.5,"value_inc_vat":5.775,"valid_from":"2018-04-17T02:30:00Z","valid_to":"2018-04-17T03:00:00Z"},{"value_exc_vat":6.82,"value_inc_vat":7.161 [... cut ...] valid_to":"2018-04-15T22:30:00Z"},{"value_exc_vat":8.36,"value_inc_vat":8.778,"valid_from":"2018-04-15T21:30:00Z","valid_to":"2018-04-15T22:00:00Z"},{"value_exc_vat":9.44,"value_inc_vat":9.912,"valid_from":"2018-04-15T21:00:00Z","valid_to":"2018-04-15T21:30:00Z"},{"value_exc_vat":9.94,"value_inc_vat":10.437,"valid_from":"2018-04-15T20:30:00Z","valid_to":"2018-04-15T21:00:00Z"},{"value_exc_vat":12.65,"value_inc_vat":13.2825,"valid_from":"2018-04-15T20:00:00Z","valid_to":"2018-04-15T20:30:00Z"}]}`

NickJenkey, 2018-10-09:
Really helpful thanks, now to incorporate it into the script to control the battery storage...

glynhudson, 2019-05-02:
Octopus also make available an API to obtain your history consumption:

See: https://octopus.energy/dashboard/developer/

e.g 

`curl -u "API KEY:" https://api.octopus.energy/v1/electricity-meter-points/1300012035212/meters/18P6806936/consumption/`

```
[{"consumption":0.087,"interval_start":"2019-03-27T23:30:00Z","interval_end":"2019-03-28T00:00:00Z"},{"consumption":0.089,"interval_start":"2019-03-27T23:00:00Z","interval_end":"2019-03-27T23:30:00Z"},{"consumption":0.423,"interval_start":"2019-03-27T22:30:00Z","interval_end":"2019-03-27T23:00:00Z"},{"consumption":1.31,"interval_start":"2019-03-27T22:00:00Z","interval_end":"2019-03-27T22:30:00Z"},{"consumption":0.117,"interval_start":"2019-03-27T21:30:00Z","interval_end":"2019-03-27T22:00:00Z"},{"consumption":0.111,"interval_start":"2019-03
```