OpenEVSE/ESP8266_WiFi_v2.x #184: Fix issue where Start/Stop button will not work when charging timer is active.

glynhudson, 2018-05-18:
As per https://github.com/lincomatic/open_evse/issues/88 there is a current issue with latest open_evse FW where the 'Start' button on the web UI will not start a charge if the delay timer is active.

A work around for this is to use a new RAPI command (`$F1`) to simulate the front LCD button press whenever the start/stop web UI button is pressed. 

The Wifi gateway should first try and send `$F1` whenever the start/stop button is pressed instead of `$FE` and `$FS`. If `NK` is returned we should fallback to using FE/FS

The new open_evse FW with F1 RAPI command it currently in testing branch and will be merged into dev branch soon. See issue above

jeremypoulter, 2018-07-02:
Just been looking at this, on my version (D4.8.0) the $FE works fine when sleeping with the delay timer. However it is when the delay timer has triggered a timer that I see an issue, the delay timer starts the charge, I click pause or press the button, the OpenEVSE briefly goes to sleep but starts again after a few seconds. 

Will update to the latest 5.0.0 firmware but we are probably going to have to do significant testing with different versions of the firmware. 

It would greatly help to list the major OpenEVSE firmware versions that need supporting, along with what happens now and what is expected (from an end user POV).

glynhudson, 2018-10-04:
Can confirm tested with [open_evse V5.0.0](https://github.com/openenergymonitor/open_evse/releases/tag/v5.0.0) FW and it works great. Fixes the issue of the start button on the web UI not being able to override the delay timer.

commit 5dc9e67e (2018-10-04): 'Fix' for changing charge mode when the timer is enabled.

Fixes #184, as detailed in the ticket there is an issue with sendinf $FE/$FS when the delay timer is set. This change detects is the time is enabled and sends the $F1 command instead to emulate a button press, and this will toggle the charging mode.