OpenEVSE/ESP8266_WiFi_v2.x #187: Implement NFC

chris1howell, 2018-06-14:
Implement NFC/RFID Read, Write and authenticate.

Add ability to read NFC tag and authorize based on a locally stored tag.

Developers tool - Add ability to write to NFC tag and store tags that are authorized locally.

**Future** OCPP Add ability to read tag and pass tag via OCPP for authentication. Accept or reject tag based on response from OCPP server.

I plan to start working on this with a common PN532 chip and PN532.h libraries over I2C (connected to Huzzah).

Example code:
[](https://www.14core.com/wiring-the-pn532-nfc-near-field-communication-devices-with-microcontrollers/)

doanerock, 2022-05-04:
I agree I would love this.  This would be great for keeping the charger from activating until the owner returns or if I want to keep track of how much a renter uses.  It is too bad that there has not been any activity in the last four years on this.

chris1howell, 2022-05-05:
The feature is available on the ESP32 version 4.x here:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/actions/runs/2045138078

ESP8266 2.x is no longer supported or under devolopment.

doanerock, 2022-05-05:
Thats amazing.  Is there any  documentation on how it works, and configured?  Are there RFID modules for sale?