OpenEVSE/ESP8266_WiFi_v2.x #190: Solar PV Divert - Create an option to stop charging when below 1.44 kW (6 A)

DIYsunnyday, 2018-08-04:
This is a proposed configuration option (e.g., a checkbox) that would enable EV-charging to periodically pause during expensive peak-demand hours when decreasing solar generation can no longer support both a home A/C system and EV charging, but it is still desirable to not send solar power to the utility where its NEM credits land in peak-demand jail. Additionally, an optional entry specifying utility peak-demand days and hours could specify when Solar PV Diversion should be active, removing EV-charging constraints during off-peak times. At the end of the day, when available solar power drops below 1.44 kW (6 A), the OpenEVSE can pause until peak-demand hours end, To keep the system stable during cloudy or A/C on periods, a configurable re-turn-on delay (e.g., 1-5 minutes) can be be implemented to prevent the OpenEVSE from oscillating between on and off.

Zuikkis, 2020-03-16:
This is from 2018 and still open. Any thoughts?

I don't own OpenEVSE yet but I was planning to, but I noticed this feature in the docs and thought it was weird.

I have a solar plant, and also market-priced electricity where each hour if differently priced. The evening hours are usually very expensive, cheaper hours start usually after midnight.

So for me, most optimal charging method would be: first with ECO using solar, and then when sun goes out, charge is stopped, and continued after midnight when bought electricity is cheap..

I understand you don't want to toggle the contactor several times a day, but for me it would be enough to automatically turn off when solar production drops below 6A, and not turn back on.. So it would be once a day.

jeremypoulter, 2020-03-17:
We are working on this currently as part of the ESP32 port: OpenEVSE/ESP32_WiFi_V3.x#54 but will be back porting. 

There is also the [demand shaper](https://community.openenergymonitor.org/t/emoncms-demand-shaper-module/9097) module for EmonCMS that handles the varying electricity charges. However I am not too sure how the two features will interact.

glynhudson, 2020-07-28:
This feature is now available for ESP8266 in V2.8.0

closed 2020-07-28 by glynhudson