OpenEVSE/ESP8266_WiFi_v2.x #193: www_password lost on reboot

jshank, 2018-08-27:
I've now twice been locked out of the web interface. I set the password as "VPIBaqzUoM70CStt" (generated by LastPass). The first time was after a firmware upgrade to 2.7.4. I just did a restart on the unit and my credentials aren't working again. Is this password too complicated? In the administration section, it shows my username (jshank) with a password of ___DUMMY_PASSWORD___.

OpenEVSE WiFi
Firmware: | 2.7.4
Flash Size: | 4096K
Free RAM: | 18K

jeremypoulter, 2018-08-27:
Hi, I am very sorry about this. 

It looks like it very well could be the password length. From what I can see the password is limited to 15 chars however we don't look to error or anything if this limit is breached. I think the password will just be cropped so should be `VPIBaqzUoM70CSt`. 

The UI can definitely be improved in regard to validation of inputs, @glynhudson what do you think?

The `___DUMMY_PASSWORD___` is just a magic value used by the UI if you have set a password, so the UI will show something in the password box but you can't view the password by using view source or the browser inspector, etc.

jeremypoulter, 2018-08-28:
It will probably have to be 15 chars in the short term, changing the length will cause the current saved password to be lost, the structure config is not really designed to be changed

glynhudson, 2018-08-28:
Yes, I agree. Let's stick with 15 characters for the moment but add validation to ensure this doesn't catch people out. Thanks for reporting @jshank :+1:

glynhudson, 2019-01-09:
Hi @Suriv , thanks for the PR. Nice idea, I've just tested but it seems the text box color stays green even if I enter a long password. I'm still able to save a long password 

[image]

Note: `DUMMY_PASSWORD` also needs to be changed in `web_server.cpp`

Suriv, 2019-01-09:
Hi @glynhudson 

If you help me compile the html and move it to the data or the simulator, I will be able to verify it.

I have done tests in the simulator.

glynhudson, 2019-01-10:
Ah ok, I was testing on the HTTP password field rather than the WiFi PSK field sinc this is what this issue is regarding. The WiFi password can be up to 64 characters long which is the maximum permissible WiFi password length therefore validation is not essentail:  https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/a5a43a9f025b00d85671e49b0e42f0fab19d173b/src/config.cpp#L36

The http username / password can only be 16 characters long, validation is needed to ensure user does not try and set a long er username / password. https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/a5a43a9f025b00d85671e49b0e42f0fab19d173b/src/config.cpp#L48

What issues are you having uploading the code? Have you got platformio up and running? Assuming you have seen: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x#firmware-compile--upload

Suriv, 2019-01-12:
update

glynhudson, 2019-01-18:
Nice work, I've just tested PR. It now works as expected which is great. Couple of issues:

- The color validation seems to be the wrong way round, the field is red until the character limit is reached, it then turns green. IMO the colours should be inverted
- The max number of characters the http authentication can handle is 15, the limit set in your PR is 16. This should be lowered to 15. 
 
[image]
[image]

Suriv, 2019-01-18:
Hello

I should have exposed the colors:
Red - Invalid field of the estimated characters.

Green- Valid field that supports all 15 characters.

Blue - I have thought for the fields that are obligatory as for example the *

The validation fields behave in the following way.

When you write the field is invalid until you complete the 15 characters that are necessary there from the red color, when it completes it becomes valid because the 15 characters have been added but it is valid from the incorrect requirement the password which is another different validation.

Now, if you tell me that the field can be 0 to 15 characters, I should change the parameter to make it work.

The configuration of the password field indicates 16, I will just change it to 15.

Where does it indicate that they are 15 characters for the user?

Suriv, 2019-01-20:
The point is that I did not get to 16 because it is not going to leave the field with the max-length property of the input that is set

I have changed the formats, since the red color would be to indicate error, I have put the blue to indicate that the field is required and it becomes green when you reach those 15 characters telling the user that it is correct.

glynhudson, 2019-01-21:
Sure, I'm happy with that.

Couple of things:

- "the field requires 15 characters" should be changed to "15 characters max"
- there seems to be a duplicate of the username field (maybe this is a browser cache issue? can you replicate)

Suriv, 2019-01-22:
It’s my pleasure

commit 012d6c82 (2019-01-24): Merge pull request #203 from Suriv/193

www_password limited to 15-characters but not validated fix #193