OpenEVSE/ESP8266_WiFi_v2.x #199: Json keys and string values need to be wrapped in double quotes

yoavl, 2019-01-01:
Services written in different languages ​​(eg golang) are very strict on Json format and will throw error if there are no double quotes around the keys

jeremypoulter, 2019-01-01:
Just had a quick check and the few I looked at look to be quoted correctly.   Please can you give the specific endpoint that has an error?

yoavl, 2019-01-02:
The Function create_rapi_json() 
In the current state the request is:
2019/01/02 12:20:14 GET /input/post.json?node=test&json={amp:0,wh:0,temp1:0,temp2:241,temp3:230,pilot:16,state:1,freeram:30616,divertmode:1}&apikey=72dle7b5d934bb0999f5oh3ca2078a96 HTTP/1.1
Host: 192.168.101.50:8081
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: close
User-Agent: ESP8266HTTPClient

I have changed the function:
create_rapi_json() {
  url = e_url;
  data = "";
  url += String(emoncms_node) + "&json={";
  data += "\\"amp\\":" + String(amp) + ",";
  if (volt > 0) {
    data += "volt:" + String(volt) + ",";
  }
  data += "\\"wh\\":" + String(watthour_total) + ",";
  data += "\\"temp1\\":" + String(temp1) + ",";
  data += "\\"temp2\\":" + String(temp2) + ",";
  data += "\\"temp3\\":" + String(temp3) + ",";
  data += "\\"pilot\\":" + String(pilot) + ",";
  data += "\\"state\\":" + String(state) + ",";
  data += "\\"freeram\\":" + String(ESP.getFreeHeap()) + ",";
  data += "\\"divertmode\\":" + String(divertmode);
  url += data;
  if (emoncms_server == "data.openevse.com/emoncms") {
    // data.openevse uses device module
    url += "}&devicekey=" + emoncms_apikey;
  } else {
    // emoncms.org does not use device module
    url += "}&apikey=" + emoncms_apikey;
  }

  //DEBUG.print(emoncms_server.c_str() + String(url));
}

Now the request is:
2019/01/02 12:30:49 GET /input/post.json?node=test&json={"amp":0,"wh":0,"temp1":0,"temp2":226,"temp3":230,"pilot":16,"state":1,"freeram":31328,"divertmode":1}&apikey=72dle7b5d934bb0999f5oh3ca2078a96 HTTP/1.1
Host: 192.168.101.50:8081
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: close
User-Agent: ESP8266HTTPClient

I checked the change in my existing account in data.openevse.com/emoncms and everything works as expected.
Now the new service I'm trying (written in golang) works great.

jeremypoulter, 2019-01-03:
Ah that JSON ;-) I _think_ that should be ok, may cause some issues with legacy setups but the current EmonCMS handes this OK, @glynhudson what do you think?

yoavl, 2019-01-03:
I also looked at the mqtt_publish() function, I think it will work fine there too.
Currently I do not have a way to check the MQTT service.

glynhudson, 2019-01-07:
Great, thanks. I've just tested and emoncms handles it just fine. PR has been merged :+1:

commit 48f0377c (2019-01-07): Resolve #199. Add Escaped Double quote to the json string
commit d7b90cd6 (2019-01-07): Merge pull request #200 from yoavl/double_quotes_for_json_keys

Resolve #199. Add escaped double-quote to the json string
closed 2019-01-07 by glynhudson