OpenEVSE/ESP8266_WiFi_v2.x #215: MQTT RAPI Response should include cmd (or at least information to route response)

TonyApuzzo, 2019-02-05:
All MQTT RAPI Responses are published to <topic>/rapi/out -- this is not enough for asynchronous clients to know what to do with the responses. 

on or around [mqtt.cpp line 83](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/master/src/mqtt.cpp#L83) could publish to <topic>/rapi/out/$XX so that clients could listen for updates to the charging current or whatever without having to heuristically determine what response format is.

In the example of $GC, it is probably not safe in general to assume that any publish of topic/rapi/out that regex matches `/$OK [0-9]+ [0-9]+ /` is the min/max current, and also there could be extra stuff to deal with like the checksum and sequence # which makes it even trickier to get right and every client has to build a bespoke router just for responses rather than using the built-in MQTT routing ability of sub-topics.

Using the sequence number ":hh" is a possible workaround but seems needlessly complicated and would not be good for interoperability.

Even the HTTP API returns both `cmd` and `ret` for the synchronous call.

TonyApuzzo, 2019-04-07:
No response, so am I missing something?  I don't understand how to tell in an asynchronous environment which response maps to what request.

glynhudson, 2019-04-08:
Recommend using HTTP for the moment if you require this functionality

TonyApuzzo, 2019-04-08:
OK, that is what I'm doing.  Unfortunately there appears to be memory leaks or some other instability in the HTTP server.  I get significant drop-outs where the HTTP RAPI times out or fails, and then it will randomly come back and then fail, etc.  

This is the result of running HTTP RAPI $GU periodically over a day -- the ~8:30pm transition to zero is legit, but the dropouts and spurious zeros are not:

[image]

paulbernardr, 2019-04-08:
I have 2 EVSE units and they fall over regularly from either http rapi or mqtt rapi communication. They do eventually come back to life, usually after 20mins or so.

paulbernardr, 2019-04-10:
The units tend to fall over (go back to ip address on display) when they are sleeping. My code is then just reading the openevse/amp feed and am not sending any commands as I am waiting for my PV to have sufficient power ie. 1500W.

glynhudson, 2019-04-10:
I've not experienced the WiFi module re-booting, what's your setup? What OpenEVSE hardware and ESp8266 hardware are you using?

paulbernardr, 2019-04-11:
Both units supplied by openenergymonitor shop with wifi modules
Firmware 5.1.0.EU
Protocol 4.0.1
Wifi 2.8.0
So both units have Esp8266 modules.

paulbernardr, 2019-04-11:
Adafruit ESP8266 Huzzah WIFI module is as sold from the openenergymonitor shop December 2018
....appreciating the WIFI is still in beta maybe this version of the ESP8266 are not up to the task?

glynhudson, 2019-04-12:
Great, that's the most recent FW version. We're running lots of test units and I've never seen the WiFi unit reboot. Maybe it's polling the unit which is causing the reboot? Could you temporally pause your poling script to see if this stops the rebooting? As I mentioned previously I would recommend subscribing the to MQTT status updates published by the unit rather than polling the unit, the ESP8266 is very resource constrained. We are working on an integrated embedded Linux (RasPi) option for more powerful applications