OpenEVSE/ESP8266_WiFi_v2.x #219: Investigate Mongoose as a replacement for AsyncWebServer

jeremypoulter, 2019-02-21:
Mostly here as a reminder...

See discussion here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/324#issuecomment-465984961

https://github.com/cesanta/mongoose

zhivko, 2019-03-31:
So is mongoose on esp32 possible? Have you tried it with websocket unctionality?

jeremypoulter, 2019-03-31:
I believe others have but I have not tried anything yet, just been following the discussions and Mongoose came up as an alternative to ESPAsyncWebServer due to the instability. If you have any prior experience with Mongoose (the web server not the full OS) them please share.

jeremypoulter, 2019-06-22:
FYI you can follow my progress on this on my fork, https://github.com/jeremypoulter/ESP8266_WiFi_v2.x/tree/mongoose_port.

Currently the webserver is basically working via Mongoose and my initial tests show it is a lot more stable than ASyncWebServer.