OpenEVSE/ESP8266_WiFi_v2.x #223: Won't start timed charging on Renault Zoe R110

snitkjaer, 2019-03-28:
When I plug in a charge my Renault Zoe right away everything works correctly.

However if I set the EVSE to do charing say 2 hours later the Zoe seems to have gone to sleep i.e. wont start charging. 

Work arounds for waking up the Zoe:
1) Press the key / remote lock (or any other button)
2) Restart EVSE

In both cases the Zoe will come back up and start charging.

This might actually be a bug in the Zoe but I guess it should be possible to implement a workaround that will wake up the car in case it does not start charging on timer.

Tested on:
OpenEVSE: 5.0.0.3P
OpenEVSE Wifi: 2.8.0

paulbernardr, 2019-03-28:
This is standard on the Zoe and some other EV's but not the Leaf. 
Car needs to charge a teeny bit before you can put the EVSE to sleep.

snitkjaer, 2019-04-01:
> This is standard on the Zoe and some other EV's but not the Leaf.
> Car needs to charge a teeny bit before you can put the EVSE to sleep.

Would be nice with a feature in OpenEVSE that would detect that charging does not start and reset the connection or something like that.

paulbernardr, 2019-04-01:
I have written a python script that interacts with an emonpi controlling the  charging. As I having a Solar PV, this means I dont pay a penny at home to charge my two cars!

glynhudson, 2019-04-01:
The Zoe is awkward, it's the only EV that I know of that behaves like this.  

RAPI command `$FR` will restart the OpenEVSE. This can be sent via HTTP or MQTT e.g 

http://openevse.local/r?rapi=%24FR

paulbernardr, 2019-04-01:
Hi Glyn
Yes for my Zoe's, I ensure that the EVSE is 'Ready' before the car is plugged in and then my script instigates a 30sec charge at 6A before sending a rapi 'Sleep' command, if the PV is not active enough. To wake the EVSE and get the charge to start, I have to send a rapi 'reset' command as an 'enable' command does nothing.

snitkjaer, 2019-04-02:
I was considering modifying OpenEVSE Wifi to add a feature (that can be enabled using a tick box) which would restart OpenEVSE when the timer elapses if a car is connected & the charging does not start within 20 seconds. I understand it can be done with NodeRed / Python / C# or whatever one prefer using the RAPI, which might be a worth considering but I normally prefer not to have to many services running for the basics to work i.e. charge the car. The wife becomes unhappy if her car i not charged in the morning :-) My reason for using timed charge is that I get billed on the Nord Pool spot price which is much cheaper at night.

snitkjaer, 2019-04-09:
Great link thanks Glyn.
It's an R110 ZE40 which I believe has the same charger as R90 ZE40. 
I normally charge with 11 kW as a normal legal house fuse i DK is 25A and I want to leave something for the washing machine etc.

paulbernardr, 2019-04-09:
I have a 2015 ZE-22KWhr and a 2017 ZE-41KWHr both R models. They both charge at 6A okay. Lower charging efficiency at lower set charge currents makes sense as a constant base input power is obviously required to energise the inverter in the first place.

ncaunt, 2019-12-31:
I recently started setting the delay timer over MQTT and found my Zoe (22 kWh R90) often would not charge when the start time came around. Sending `$FR` later initiates a charge (sometimes needing more than one attempt) but I've also found sending `$FD` then `$FE` works. I've not tried running a short charge before sleeping yet but I wondered if the delay timer problem is related to the need for this workaround: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/76577408ca1ce8d8266db463adfe5f0b173b1d86/src/divert.cpp#L209

jpalo, 2020-03-25:
@snitkjaer : I've sent a [PR](https://github.com/emoncms/demandshaper/pull/42) adding support for Nordpool Spot in Demand Shaper module. Been using it few months with success.

glynhudson, 2020-09-14:
We have added the feature to V3.3.1 under advanced options to use Disable instead of Sleep https://github.com/OpenEVSE/ESP32_WiFi_V3.x/releases/tag/3.3.1

V3 firmware requires an ESP32 module: https://shop.openenergymonitor.com/openevse-wifi-gateway-v3-x-esp32/

[image]

closed 2020-09-14 by glynhudson

KipK, 2023-01-05:
I'm waking up this thread, but I got some users reporting me that Pause Status set to Disable still set EVSE in sleep mode ( 254 ) and not Disabled ( 255 ) on current firmware.
Does somebody have this worlking  ?