OpenEVSE/ESP8266_WiFi_v2.x #224: Implement schedules for days / times?

Edison517, 2019-04-10:
I apologize if this request is being worked on already, but if so, I couldn't find it.

I'd like the ability to set schedules for charging by day, and time, multiple times per day (if possible).

I have time-of-use billing and I would rather have the schedule of cheap rates (mostly at night) done on my OpenEVSE rather than my 2014 Volt. I know that right now I can set a daily schedule, but that doesn't help me when my weekend rates are the same as my nightly rates during the week :)

glynhudson, 2019-04-12:
Yes, being able to set a weekly charging schedule is something we will be working on implementing in the near future.

glynhudson, 2019-04-30:
[image]

Possible schedule UI 

https://github.com/openenergymonitor/ESP8266_Relay_Board/blob/master/firmware/html/control/thermostat.js
https://github.com/openenergymonitor/ESP8266_Relay_Board/blob/master/firmware/html/control/thermostat.html

Maybe keep the bars to visualise each day or the week but set the on/off time via a browser time picker. Android automatically then uses the nice inbuilt time picker