OpenEVSE/ESP8266_WiFi_v2.x #226: New UI Memory Usage

glynhudson, 2019-05-10:
This issue is to track / record the increase in data / program memory required 

Old: https://github.com/OpenEVSE/openevse_wifi_gui/commit/7036fe9cc9fe5504270ee6a870061fb5047a7a8a
New: https://github.com/Suriv/openevse_wifi_gui/commit/207d4ec3425511219ffa816954ff559a294f9779

[image]

jeremypoulter, 2019-05-10:
So ~100kB increase, that is probably acceptable.

glynhudson, 2019-05-12:
Seems to be working fine on the ESP8266, been running it on my home OpenEVSE over the weekend, loading time is a bit slower though, but just about acceptable. See video (loading graphic still need some work) https://drive.google.com/file/d/1cYGzEHU0iaeAwBxsqUun7kfyDvDXbsD0/view

Pukkahq, 2019-05-24:
@glynhudson 

That looks a little slow, I access the web page on a regularly on my mobile device to turn on eco charging and then to stop charging one we are producing < 6A. can these be settable default options.

glynhudson, 2019-05-24:
I agree, this new UI is in early stage development. We will ensure performance is higher before release.

closed 2020-07-28 by glynhudson