OpenEVSE/ESP8266_WiFi_v2.x #230: Wifi Scan results -> Invalid Json

adgoudsmit, 2019-05-27:
When i tried to setup my AP to be used, i found out there was no wifi AP displayed on the website.
I eventually found out one of my neighbours has an AP with " in the name which caused the json to be invalid for the website to display the list of available accesspoints.

Version used: 2.8.1

glynhudson, 2019-05-27:
Thanks for reporting, that's very useful to know. Should be an easy fix

jeremypoulter, 2019-05-27:
Well, that is just plain irritating, who has " in their SSID ;-) 

@glynhudson any objections to using [ArduinoJson ](https://arduinojson.org/) rather than handcrafting the JSON? Will automatically handle cases like this.

glynhudson, 2019-05-27:
Haha agreed. It's bound to cause issues with other devices. Sure, that sounds like a good idea. 

At the same time it would be useful to implement some or all of:  https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/204

chris1howell, 2019-05-27:
Sounds like a great plan. Should clean things up a lot.

fiveangle, 2019-05-29:
> Well, that is just plain irritating, who has " in their SSID ;-)

As SSIDEncoding field can be UNSPECIFIED (for arbitrary data) and UTF8, yes, even 32 NULL characters would be valid so using any string operations is a no-no

SSID: (╯°□°)╯︵ ┻━┻

closed 2020-07-28 by glynhudson