OpenEVSE/ESP8266_WiFi_v2.x #231: A few Solar PV divert issues

jbakuwel, 2019-06-15:
Solar PV divert only works for Service Level 2 and currently needs a positive value (not a negative value as described in the documentation) in the SolarPV-gen topic to adjust the charging current. The calculation seems to be based on 110VAC (despite the fact that Service Level =  2) as publishing 770 in the SolarPV-gen topic results in 7A, 880 in 8A and so on.

I am not sure why Solar PV would/should not work for Service Level 1 but if it's not meant to work for Service Level 1 then perhaps Charge Mode should be greyed out when Service Level is set to 1.

Grid (+I/-E) topic is greyed out (ie. I can't define that topic). In my case (off grid) that is fine of course but I thought I'd mention it.

As also mentioned in now closed issue #227, it would be great for off grid systems if Eco mode would be persistent across charging sessions (ie. remain set to Eco after unplugging and replugging the car).

Instead of using a contactor, using a zero cross SSR such as https://www.digikey.com/product-detail/en/sensata-crydom/D2440/CC1071-ND/221764 might allow the firmware to pause & resume charging more frequently (probably with some form of hysteresis) - assuming this would not wreck havoc on the car charger. Zero cross SSR might help. Heat dissipation might be an issue for the current casing though... just an idea :-)

glynhudson, 2019-06-26:
>The calculation seems to be based on 110VAC (despite the fact that Service Level = 2) as publishing 770 in the SolarPV-gen topic results in 7A, 880 in 8A and so on.

This has been fixed in [V2.8.1](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.8.1)

>I am not sure why Solar PV would/should not work for Service Level 1 but if it's not meant to work for Service Level 1 then perhaps Charge Mode should be greyed out when Service Level is set to 1.

Solar PV divert will only work for Level 2 since calculation is based on L2 voltage. Good point, it should be grayed out for L1 

>Grid (+I/-E) topic is greyed out (ie. I can't define that topic). In my case (off grid) that is fine of course but I thought I'd mention it.

If you remove the entry in solar PV gen topic the Grid I/E field will become usable instead. It's either / or. Agree, this could be make clearer with a toggle. 

>As also mentioned in now closed issue #227, it would be great for off grid systems if Eco mode would be persistent across charging sessions (ie. remain set to Eco after unplugging and replugging the car).

This would have to be an option, since most users would prefer EVSE to revert to Normal mode to avoid being stranded with a low SoC if full charge speed was expected. 

> Instead of using a contactor, using a zero cross SSR such as https://www.digikey.com/product-detail/en/sensata-crydom/D2440/CC1071-ND/221764 might allow the firmware to pause & resume charging more [...]

A 2P contactor is required by regulations. It's actually the contactor inside the EV that I would be more worried about wearing out by stopping / starting often. The contactor in the EVE is easy to replace.

jbakuwel, 2019-06-26:
Hi Glyn,

> On 27/06/2019, at 5:24 AM, Glyn Hudson <notifications@github.com> wrote:
> 
> The calculation seems to be based on 110VAC (despite the fact that Service Level = 2) as publishing 770 in the SolarPV-gen topic results in 7A, 880 in 8A and so on.
> 
> This has been fixed in V2.8.1
> 

Great, thanks. It seems though the adjustment of the amps is buggy (at least in v2.8.0) or perhaps more changes have been made in v2.8.1? I'm looking at a screen here with max current set to 9A, published solar power of 2KW, yet charging remains fixed at 6A. Sometimes disabling PV divert and setting max current to say 9A has no effect, with Amps sticking to a lower value, while I'd expect to see the Amps rise to the max level set.

While I'm quite able to hook up a serial console, it's quite inconvenient as the car is outside, meaning I'd need to run a very long serial cable and also leave the box open/exposed. Being able to get debug messages via the network would be a lot more convenient.

Would it be possible to have the ESP8266 (in developer mode) log debug messages to a syslog server?

> I am not sure why Solar PV would/should not work for Service Level 1 but if it's not meant to work for Service Level 1 then perhaps Charge Mode should be greyed out when Service Level is set to 1. [...]

Would it be even better to use the service level for the calculation?
Ie. when set to level 1, use 110V and when using level 2 use 240V?
I'd assume PV divert would also be useful for folks in countries with a 110VAC mains?

> Grid (+I/-E) topic is greyed out (ie. I can't define that topic). In my case (off grid) that is fine of course but I thought I'd mention it.
>
> If you remove the entry in solar PV gen topic the Grid [...]

Indeed.
> Instead of using a contactor, using a zero cross SSR such as https://www.digikey.com/product-detail/en/sensata-crydom/D2440/CC1071-ND/221764 might allow the firmware to pause & resume charging more [...]

That makes sense, thanks.

Jan

nitroushhh, 2019-06-30:
Regarding to the suggestion to make charge mode stick at ECO.  I agree that having it always set to eco would be a problem.  Would it be possible to have a field where we can pick what the default is.  That would suit me well as in a sunny week default 'eco' would be good.  In winter I could set default to 'normal'

beikeland, 2019-12-01:
I tried changing the default to ECO, but it also needs OpenEVSE to default to sleeping for it to not start charging if turned on with car connected when solar is insufficient. Quite possible I overlooked something however.

glynhudson, 2020-07-28:
solar PV divert has been significantly improved in V2.9.0

closed 2020-07-28 by glynhudson