OpenEVSE/ESP8266_WiFi_v2.x #232: API documentation

jeremypoulter, 2019-06-22:
We should write some API documentation on the HTTP API, would probably be a good idea to use [OpenAPI](https://www.openapis.org/) and [Swagger](https://swagger.io/)

glynhudson, 2019-06-27:
Definitely a good idea to improve docs.

I'm not sure a 3rd party service is required. While, I agree those services do make very nice API docs I don't think the API is complex enough to merit their use. I also worry about the longevity and maintenance over time. My preference would be a simple markdown file or addition to readme.md.

chris1howell, 2019-06-28:
Here is what I have done to date. It would be a good place to start.

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/tree/master/Developers_Guides

jeremypoulter, 2019-07-18:
Using OpenAPI is not just about pretty docs, it opens up a lot of scope for testing and auto-generated SDKs: https://openapi.tools/