OpenEVSE/ESP8266_WiFi_v2.x #237: Add Support for RAPI 5.0.0

lincomatic, 2019-07-10:
I added hooks to facilitate support of OCPP. In the process, RAPI got altered a bit, and became incompatible with existing code. The changes are as follows:

1. $ST async notification replaced with $AT, which has more parameters
2. $AB async notification is now sent on boot
3. $GS has 2 new parameters added to it. Also, the first parameter, state, is now in hexadecimal, instead of decimal, to make it consistent with $AT (and the now deprecated $ST)
going forwards all async new async notifications will be of form $Ax.

I was going to create a pull request, but then realized that I don't understand your code well enough to fully test it. I've put some changes into the following commit in my own branch:

https://github.com/lincomatic/ESP8266_WiFi_v2.x/commit/da56fea96a67225d8d3bb08e3b618207b5ba97b1

Please review them, and alter as needed, at your convenience.

jeremypoulter, 2019-07-19:
@lincomatic I will indeed look at changing this. Does this include the support for detecting if the car is connected/disconnected in sleep mode? If not could it as these are kind of breaking changes. would be good to also get this in there.

lincomatic, 2019-07-22:
> @lincomatic I will indeed look at changing this. Does this include the support for detecting if the car is connected/disconnected in sleep mode? If not could it as these are kind of breaking changes. would be good to also get this in there.

Yes, it does. It's one of the vflags bits

#define ECVF_EV_CONNECTED       0x0100 // EV connected - valid only when pilot not N12

returned by the new extended $GS

GS - get state
 response: $OK evsestate elapsed pilotstate vflags
 evsestate(hex): EVSE_STATE_xxx
 elapsed(dec): elapsed charge time in seconds of current or last charging session
 pilotstate(hex): EVSE_STATE_xxx
 vflags(hex): EVCF_xxx
 $GS^30

beikeland, 2019-12-06:
When testing the OpenEVSE D6.0.0  FW which uses RAPI 5.0.0 its clear that the Wifi FW does not verify the RAPI version reported by OpenEVSE FW  to determine compatability but rather breaks silently. 

$GV is used to display the `protocol` variable, so a check to determine compatibility and display a warning would be a nice feature.