OpenEVSE/ESP8266_WiFi_v2.x #239: Captive Portal not working with Google Pixel 3

jeremypoulter, 2019-07-28:
Just tried to setup using my Google Pixel 3 and the captive Portal does not load.

This is probably more of an Android 9 issue than just a Pixel 3 issue.

glynhudson, 2019-08-01:
Interesting, what's the failure mode? Do other captive portals work on the Pixel3?

jeremypoulter, 2019-08-01:
Basically nothing loaded in the captive portal window and if you go directly to Chrome and enter the address it timesout

jeremypoulter, 2019-08-03:
This may actually be more of a general problem with the Pixel 3/Android 9/Mobile Chrome, can not get the UI to load at all even in normal WiFi client mode.

Works ok on desktop Chrome and setup fine for MQTT/EmonCMS so definitely not a network issue, but the UI just never loads on my mobile.

glynhudson, 2019-10-09:
Should this now be fixed by https://github.com/OpenEVSE/openevse_wifi_gui/pull/16? Could you test ESP8266 on captive portal on you pixel3? @jeremypoulter

jeremypoulter, 2019-10-10:
I don't think so, I think this issue is related to the web server not the GUI. Works on the ESP32 as that is using Mongoose.

closed 2020-07-28 by glynhudson