OpenEVSE/ESP8266_WiFi_v2.x #243: WiFi interface not loading on iphone 7?

glynhudson, 2019-08-19:
It's been reported that the interface failes to load on an iphone 7, see: 
https://drive.google.com/file/d/1S6-xt_8SqLawJKijID9nlLctJYaOSgYU/view

I've been able to verify that it works fine on an iphone 6, is someone able to test with an iphone 7?

digitalarchitects, 2019-08-19:
I have a 7

https://drive.google.com/file/d/1S6-xt_8SqLawJKijID9nlLctJYaOSgYU/view

glynhudson, 2019-08-19:
Thanks, does the OpenEVSE WiFi web interface load on your iphone 7?

digitalarchitects, 2019-08-19:
I would need to setup a web interface. I will chat to Marcus to get involved.

regards,

Kevin Wilson

+27 11 552 4093 (Office)
+27 82 901 4224 (Mobile)
+27 86 676 6065 (DigiFax)

Stefanutti Stocks Corporate Services
a division of Stefanutti Stocks (Pty) Ltd
Reg No. 2003/022221/07

[https://intranet.stefstocks.com/_layouts/images/exclaimer/logo.png]

Suriv, 2019-08-20:
I have done the test and it works for me.

Can there be two devices connected to the same address or only one connection is allowed?

In my opinion, a news website is not the same as a configuration panel, I think it is good that only one device is connected because two devices at the same time would be changing parameters which would cause the processes to go crazy.

@digitalarchitects I don't understand why you would have to configure a web interface?

glynhudson, 2019-08-20:
>I have done the test and it works for me.

Thanks, that's good to know. Where you using the standard Safari browser and running IOS 12.4?

>Can there be two devices connected to the same address or only one connection is allowed?

It's allowed but the ESP is resource limited, therefore multiple connections may cause performance issues.

Suriv, 2019-08-20:
> Thanks, that's good to know. Where you using the standard Safari browser and running IOS 12.4?

Yes, but I've only done with a device not with two as it appears in the image.
What version of iOS does the iPad have?

glynhudson, 2019-08-20:
>What version of iOS does the iPad have?

Also OS 12.4

BrianDrury, 2019-08-21:
I am the guy with the iPhone 7 problem.

The problem was apparent when I was only using the iPhone. I checked to see if my wifes iPhone worked OK but it gave the same result so I tried our iPads but they work just fine.

Next I returned to using just the one iPhone and cannot see the full OpenEVSE page. I just get the first bit as you can see in the video.

I am happy to try any tests you can think of.

Suriv, 2019-08-22:
Hi @BrianDrury,

I did it on an iphone 7 but it is possible that my configuration is different from yours.

First test, visit with your iPhone 7, this page: https://openevse.openenergymonitor.org

It is a "functional" portal to teach functionalities. It works for you?

@glynhudson  Which version is uploaded to the address https://openevse.openenergymonitor.org?If it is an earlier one, could it be updated to the last one that is the one supposed to have the fault?

glynhudson, 2019-08-22:
>@glynhudson Which version is uploaded to the address https://openevse.openenergymonitor.org?If it is an earlier one, could it be updated to the last one that is the one supposed to have the fault?

It's the current version, but it's being served via NodeJS and Apache therefore is very different to the ESP8266 embedded web server. The demo site is more designed as a way for people to checkout the interface.

BrianDrury, 2019-08-22:
Thanks @Suriv Javascript was OFF. I turned it on and the iPhone connected immediately. I shall change the other iPhone and Ipads.

closed 2019-10-08 by glynhudson