OpenEVSE/ESP8266_WiFi_v2.x #248: Missmatch between code and comment for intervalls

beikeland, 2019-12-01:
https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/76577408ca1ce8d8266db463adfe5f0b173b1d86/src/mqtt.cpp#L187-L188

Not sure if comment or code is correct, but 5s != 50000ms

glynhudson, 2019-12-07:
This looks correct to me, try and reconnect continuously for 5s `(now < 50000)` then try to re-connect every 10s `((now - lastMqttReconnectAttempt) > 100000))`

closed 2019-12-07 by glynhudson

beikeland, 2019-12-07:
5 seconds is 5,000ms not 50,000ms which explains my trouble with mqtt thinking it was the server not accepting the connection, but instead it was the wifi waiting almost two minutes to reconnect, despite thinking it should take 10 seconds.

glynhudson, 2019-12-07:
Damit, I was obviously blind to the number of zeros! Thanks, I've made the changes to fix

commit 9ff82772 (2019-12-07): fix #248