OpenEVSE/ESP8266_WiFi_v2.x #250: Inability to set MQTT clientID

denisfrench, 2019-12-08:
I'm happily using the Losant platform (which includes a cloud-based MQTT server) as the basis of my homebrew home automation setup.

The Losant MQTT protocol requires the client ID to be sent with every connection (https://docs.losant.com/mqtt/overview/#authentication).

I do not yet own an OpenEVSE, but it is not clear to me where within the UI or code the clientID can be set.

I'm pretty sure there are other MQTT implementations that are similar (eg Cayenne).

beikeland, 2019-12-08:
https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/master/src/mqtt.cpp#L101
https://pubsubclient.knolleary.net/api.html

denisfrench, 2019-12-08:
Ahh, I see.

So it always connects to the MQTT server using a clientID which is hard-coded to a unique ID derived from the MAC address.

Recompiling with the Losant-specified clientID is simple enough, but I suggest being able to specify the clientID within the UI would be a useful feature.

beikeland, 2019-12-08:
Maybe you the newly configurable hostname could be applied here as well?

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/981495188064f19b2fe641eb0547d9af8f7ed3f0

glynhudson, 2020-07-28:
>Maybe you the newly configurable hostname could be applied here as well?

Configurable hostname is now available in V2.9.0 release

closed 2020-07-28 by glynhudson