OpenEVSE/ESP8266_WiFi_v2.x #253: Where is the code for release 2.8.2?

lincomatic, 2020-01-21:
Both the included code in the 2.8.2 release, and the current master code build a different version.
platformio.ini tags them as 2.8.1.
The firmware.bin of the 2.8.2 release is tagged as 2.8.2. Updating firmware from the 2.8.2 firmware.bin behaves differently from any of the available code .. there's a horizontal bar that grows as the update progresses.

jeremypoulter, 2020-01-21:
The 2.8.2 code is tagged with `2.8.2` here: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/tree/2.8.2 and also avalible as a snapshot on the releases page, https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.8.2

fiveangle, 2020-01-21:
@lincomatic - it seems pretty clear that DBUILD_TAG in platformio.ini had mistakenly not been bumped for 2.8.2:

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/41f9a325e459739aebda5c97b3d856b472ba7456#diff-237285c047dad808f73698c139144e1aR33

I typically submit a PR to resolve obvious oversights, or at least submit a review comment on the commit (which I have now done: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/41f9a325e459739aebda5c97b3d856b472ba7456#r36903724 )

lincomatic, 2020-01-22:
It's not only the build tag that hasn't been bumped. The code doesn't generate the same firmware. Try doing a firmware update w/ the firmware.bin that's tagged 2.8.2. And then build either the current master code or the src that's bundled w/ the release. It behaves differently. The firmware.bin has an inline horizontal progress bar as the update progresses, while any of the available code builds don't

jeremypoulter, 2020-01-22:
Yes I think maybe @glynhudson did not commit some files after building the 2.8.2 release, the code for the UI update is [here](https://github.com/OpenEVSE/openevse_wifi_gui/pull/16), but I do not see a corresponding commit of the 'built' UI files, or as pointed out the, the version number bump.

closed 2020-07-28 by glynhudson