OpenEVSE/ESP8266_WiFi_v2.x #254: create a docker file to setup latest firmware build

Niko-La, 2020-01-24:
Having trouble building a modified gui into a firmware.  
Perhaps a dockerfile would make it easy for other ppl to get up and running knowing the platformio build and other settings are just as the developers intended?

jeremypoulter, 2020-01-24:
PlatformIO normally does a fairly good job of presenting a consistent build environment so I am not too sure a Docker will help. That being said there are a few existing Platform IO docker images (eg https://github.com/suculent/platformio-docker-build) maybe try one of those.

It would also be good to get some specifics about the issues you are facing, so maybe rais another ticket about them.

Niko-La, 2020-01-29:
@jeremypoulter 

I forked it and made a few changes to the web gui header from OPENEVSE to KING HS.LA OPENEVSE

ran `npm run build`

and build a new firmware.bin using platform.io

I have pushed it to my forked repo at https://github.com/Niko-La/ESP8266_WiFi_v2.x

My serial debug is as follow ( AP is not showing up on any local brwoser)

`
============================================================ [SUCCESS] Took 52.39 seconds ============================================================
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

OpenEVSE WiFI 9402738
Firmware: 2.8.1.dev
Free: 32232
Loading config
Got '' 7 == 128 @ 0:32
Using default ''
Got '␏␇␃␏␅␂␋␃␂␋␃␂␌␔��␎�RNJD@8␅
␎␕␙"���␅
␒����' 7 == 215 @ 32:64
Using default ''
Got '␂' 2 == 130 @ 542:33
Using default ''
Got '' 1 == 128 @ 128:45
Using default 'data.openevse.com/emoncms'
Got ',' 1 == 172 @ 173:32
Using default 'openevse'
Got '␂' 255 == 130 @ 438:60
Using default ''
Got 'd␂d␃d␄d' 124 == 133 @ 205:45
Using default 'emonpi'
Got '␁|␂|␃|␄|␅|	|
|␘|␙|␚|␛|␜' 124 == 224 @ 250:32
Using default 'openevse'
Got '␝|8|9|:|;|<|=|x|y|z|{|||}|' 255 == 225 @ 282:32
Using default 'emonpi'
Got '' 255 == 128 @ 314:64
Using default 'emonpimqtt2016'
Got '' 0 == 128 @ 378:30
Using default ''
Got '' 49 == 128 @ 408:30
Using default 'emon/emonpi/power1'
Got '␁' 251 == 129 @ 498:15
Using default ''
Got '�␁' 0 == 127 @ 513:15
Using default ''
Got '' 2 == 128 @ 575:32
Using default 'openevse'
Got '��␁' 254 == 129 @ 528:10
Using default ''
Got '010002'  47 == 131 @ 538:4
Using default '000000'
After config_load_settings: 32136
Starting AP
AP IP Address: 192.168.4.1
After wifi_setup: 29632
Server started
After web_server_setup: 25944
After ota_setup: 25320
last_state = -1
state = 0
divertmode = 1
OpenEVSE not responding or not connected
>> Slow loop 1005ms
OpenEVSE not responding or not connected
>> Slow loop 506ms

`

Dont want to test this firmware on my working openevse charger.  

Is there anyreason why the gui is not showing up. The build and upload looks sucessfull and the serial output seems ok.

jeremypoulter, 2020-01-29:
You are in access point mode, when in this mode `wifi_portal.html` is loaded not `home.html`. I think that is why you do not see any changes.

Niko-La, 2020-01-29:
@jeremypoulter  hmm I don't see anything loading up. I left the wifi_portal.html unchanged

jeremypoulter, 2020-01-29:
Are you connected to the OpenEVSE wifi network so you can reach the web server?

Niko-La, 2020-01-29:
Great once i did a reboot found the AP.

I was able to get GUI up and connect to MQTT but I guess because the OpenEVSE is not connected it doesnt respond to any RAPI commands?

>> Slow loop 505ms
OpenEVSE not responding or not connected
MQTT received:
Topic: nikolaevse/rapi/in/$GV
Payload: 1580283007020
Processing as RAPI
ws[/ws][2] text-message[10]: 
>> Slow mqtt_loop 513ms
>> Slow loop 1018ms
OpenEVSE not responding or not connected
>> Slow loop 504ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
>> Slow loop 505ms
OpenEVSE not responding or not connected
>> Slow loop 505ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
>> Slow loop 505ms
OpenEVSE not responding or not connected
>> Slow loop 505ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
>> Slow loop 505ms
Looking for /status
GET
 http://10.0.0.175/status
_HEADER[Host]: 10.0.0.175
_HEADER[Connection]: keep-alive
_HEADER[Accept]: application/json, text/javascript, */*; q=0.01
_HEADER[X-Requested-With]: XMLHttpRequest
_HEADER[User-Agent]: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/79.0.3945.130 Safari/537.36
_HEADER[Referer]: http://10.0.0.175/
_HEADER[Accept-Encoding]: gzip, deflate
_HEADER[Accept-Language]: en-US,en;q=0.9
lastUpdate = 0
millis() = 921933
(millis() - lastUpdate) / 1000 = 921
OpenEVSE not responding or not connected
>> Slow loop 505ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
Time1
nikolaevse/amp = 0
nikolaevse/wh = 0
nikolaevse/temp1 = 0
nikolaevse/temp2 = 0
nikolaevse/temp3 = 0
nikolaevse/pilot = 0
nikolaevse/state = 0
nikolaevse/freeram = 22712
nikolaevse/divertmode = 1
>> Slow mqtt_publish 9ms
>> Slow loop 519ms
OpenEVSE not responding or not connected
>> Slow loop 512ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
>> Slow loop 505ms
OpenEVSE not responding or not connected
>> Slow loop 505ms
ws[/ws][2] text-message[10]: 
OpenEVSE not responding or not connected
>> Slow loop 505ms
$GS^30
--- exit ---

jeremypoulter, 2020-01-29:
Correct, without an OpenEVSE connected you will not be able to do any RAPI stuff.

Niko-La, 2020-01-30:
I notice this webserver is loading a lower is this because the EVSE is not present.

I am still able to get gui up just taking longer.

beikeland, 2020-01-30:
Presume you meant loading slower. Yes, the UI is trying to send 16(?) RAPI
commands to the OpenEVSE, and with no response they will timeout before the
UI is loaded. So you can view the UI, but anything that relies on the state
of the OpenEVSE will not be shown correctly.

closed 2020-07-28 by glynhudson