OpenEVSE/ESP8266_WiFi_v2.x #255: possible to use USB serial to interact with ESP chip

Niko-La, 2020-01-24:
I am using a Feather HUZZAH, looks like the build and firmware uploaded using platformio as intended, however I do not see the web-server on FING app.

I have not updated my main OPENEVSE setup, this is just  a back up board to see if it works properly.

Anyway to debug why the web-server is not showing up?

jeremypoulter, 2020-01-24:
You can use the `openevse_dev` environment (see https://github.com/suculent/platformio-docker-build) to get debug out, noisier areas of debug are conditionally enabled, eg for web server add `-DENABLE_DEBUG_WEB` to the `debug_flags` section in `platformio.ini`.

On a side not the Feather Huzzah (or any ESP8622 dev board with buildit UBS serial) is not particularly good for production use as you need to connect the main serial to the OpenEVSE board, should be ok for dev though.

Niko-La, 2020-01-29:
Can you elaborate on why a Bultin USB esp8622 dev board is not good for production?

 I did change the `platform.ini`  -DDEBUG_PORT=Serial -DENABLE_DEBUG_WEB 
from `Serial1` to `Serial`

Is there any succesfull implenetation of  any with LTE-M board with ESP866.  I would like to access it using some static ip so i can debug better when not on the same wifi.  Perhpas this is better asked at an ESP8266 forumn

jeremypoulter, 2020-01-29:
On the ESP8622 `Serial` is connected to the USB so you may not be able to connect it to the OpenEVSE. `Serial1` can not be used as that is Tx only there is no Rx, ok for debug, but not communicating with the EVSE.

beikeland, 2020-01-29:
While I haven't tried the ESP32 build in the 3x repo, the ESP32 would be
better suited for using software serial if you need more serial. Struggling
to find a ESP32 in the same formfactor as the Adafruit boards + shipping
and import is just not economical.

jeremypoulter, 2020-01-29:
Correct, the ESP32 is a much better board and you do not have the same issues.

https://www.adafruit.com/product/4172 is a drop in replacement for the ESP8622 but I agree is a challenge getting them at a reasonable cost. The NodeMCU32 is kind of ok, does not have the 5v compatible serial of the Huzzah but you can mostly get away with it.

Niko-La, 2020-01-30:
https://www.adafruit.com/product/3619

Although this feather huzzah is pricier.

A similar one from Aliexpress should be about halfprice.

This one has 3 hardware uart.  This should be ok correct?

I thought have a built in USB would make improve the EVSE allowing for quickly debugging via USB cable and even uploading firmware.

beikeland, 2020-01-30:
USB is nice and all, but I think the reason the non feather version is used, is that is comes with a level translator for 5v to 3.3v. Firmware updates are done over the air, for the one I wanted debugging I first used a HC05 bluetooth module, but switched to ESP-Link running on a board linked below, so no need for physical access unless something gets properly broken by changes.

ESP32 with level translator is tricker to find on Ali/eBay, I used [this for esp8266 ](https://www.aliexpress.com/item/33000086175.html) and will probably stay with it until something is available in the same form factor and allows me to chose modules modules with internal or external antenna as needed, but I digress.

Niko-La, 2020-01-31:
@beikeland aha makes sense too bad ESP32 RX/TX are not 5V tolerant.
From the Connection Picture looks like  ESP32 RX/TX are the only pins that need level shift? 
A bi-directional level shifter is two resistors and a transistor so not much of an issue for me breadboard,

The Display and its I2C looks like their are running on the Openevse and are fine as is.  Althought putting those wires through a level shifter and using this TTGO 
Diplay looks like a nice way to reduce some parts.
https://www.aliexpress.com/item/33048962331.html

@jeremypoulter great discussion happy to continue and you can close this issue if you woulf like,

beikeland, 2020-01-31:
Just beware the board you linked to does not seem to have the usb serial broken out to the pins.

For level shifting I decided to use this https://www.aliexpress.com/item/4000132503942.html provides isolation as well as translation, and works pretty well inline on a 4pin connection between the two boards. Isolate with heatshrink or 3d print and you don't need to worry about mounting or breadboarding.

closed 2020-02-03 by jeremypoulter

Niko-La, 2020-02-12:
@beikeland @jeremypoulter 

How would I connect two or more EVSE board to an ESP module? Has anyone successfully daisy chained?

jeremypoulter, 2020-02-12:
In theory you could connect up to 3 to an ESP32 as that has 3 ports or possibly more if connected via I2C (but I have never tested that). 

However, the ESP WiFi firmware only supports a single EVSE and it would be a reasonable amount of effort to support multiple EVSEs. 

I did however design the Node.JS version with multiple EVSEs in mind so it is a lot more practical to support multiple EVSE modules from that using a raspberry Pi or similar, but still needs work.

paulbernardr, 2020-02-12:
> @beikeland @jeremypoulter
> 
> How would I connect two or more EVSE board to an ESP module? Has anyone successfully daisy chained?

I control 3 OpenEVSE's with a PiZero connected to an LCD hat showing the EVSE status and monitoring microgeneration for smart charging. This works really well and may be a possible solution.

Niko-La, 2020-02-12:
@jeremypoulter I think it would be a lot of work to do uart to i2c.  It would be nice if the evse board did i2c along with uart as that would eliminate a lot of the bottlenecks. 

I will have to look at your nodejs code for the multiple evse implementation, but i supposed you would need device capable of running nodejs.  What would be the connection between the openevse and nodejs?

@paulbernardr interesting setup, how are the OpenEVSE boards connected to the pizero?