OpenEVSE/ESP8266_WiFi_v2.x #257: Consumption pr car / vehicle id detection

snitkjaer, 2020-02-19:
I have 2 electric cars sharing the same OpenEVSE charger one is a Renault Zoe R110 and the other is a Tesla Model 3.

I would like to know which car is currently connected / charging to automatically record the consumption pr car.

Is there any way to detect which car is currently connected in OpenEVSE? Can some kind of vehicle id be detected?

beikeland, 2020-02-19:
No id is sent from the car. The TM3 does however have a pretty good API to access charge state (although polling it may prevent the car from sleeping and causing higher battery usage when idle)

snitkjaer, 2020-02-19:
> No id is sent from the car. The TM3 does however have a pretty good API to access charge state (although polling it may prevent the car from sleeping and causing higher battery usage when idle)

Okay but I'm just wondering how Teslas can automatically charge at supercharges without doing anything but plugging in? Do they have a proprietary extension or something?

beikeland, 2020-02-19:
> Do they have a proprietary extension or something?

Yes, the J1772 standard does allow to replace the analouge PWM pilot with digital communication, but afaik only Tesla use this and its in any case not supported by OpenEVSE

closed 2020-07-28 by glynhudson