OpenEVSE/ESP8266_WiFi_v2.x #261: v2.9.1 issues

cecil-t, 2020-08-02:
I have 4 OpenEVSE units, and I just upgraded two of them to v2.9.1.  Those two units then wouldn't connect from my own custom web app, I received a 401 error.  Logging into the units interface directly I also noticed that my Emoncms connections were also showing a 401 error in the OpenEVSE GUI, and I checked Emoncms directly and the feeds had stopped receiving data from those two OpenEVSE units, while the other two continued working OK.

I reverted to v2.8.2 on both units and the problems went away, all functionality returned on my web interface and Emoncms began receiving data again.  I haven't had a chance to dig into the problem in more detail.

jeremypoulter, 2020-08-02:
Does your custom app mirror the EmonCMS API? IIRC we changed to use the `fulljson` API as it is more reliable and fits in better with the new internal structure.

cecil-t, 2020-08-03:
Sorry, for clarity those would be two independent problems.  My web app couldn't authenticate with the HTTP auth login on the OpenEVSE and it was getting the 401 error.  My web app is just a multi-unit user interface so I can see and interact with all 4 units on one page.

Separately, the OpenEVSE to Emoncms was also not working, completely independent of my web app.  On the "Services" tab under Energy Monitoring at the bottom near the Save button is where I saw a 401 error for that, and the Emoncms feeds had stopped receiving data.  

I didn't make any configuration changes but simply reverted to the 2.8 firmware and everything worked again as expected.

jeremypoulter, 2020-08-03:
Some other users have reported that the EmonCMS write key is lost after upgrade, can you check if this is the case? If you upgrade one unit and re-enter the API key see if that works, same with the admin password.

**Note** saving any setting will mean if you revert back to a previous build you settings will be lost.

jeremypoulter, 2020-08-03:
@glynhudson have you seen many reports of the EmonCMS key being lost during upgrade (other than https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/b87bb42f7e903b5057b22fa072a5ffed1c50352a#commitcomment-41155141)?

glynhudson, 2020-09-14:
Yes, I have head reports of Emoncms key being lost during the update.

cecil-t, 2020-10-04:
I believe the administrator http-auth username / password is also lost during the upgrade, that was the first part of my problem.