OpenEVSE/ESP8266_WiFi_v2.x #262: Unable tu use Disable EVSE mode while  PV divert is ON at 2.9.1

ayasystems, 2020-09-28:
Hi

If you have enable PV divert you can not use Disable RAPI command $FD. You can use it but if you put it to disable it is going back to charge if has FV available so you can not stop the charge while FV is available

Best regards

Bisha3, 2020-09-28:
> Hi
> 
> If you have enable PV divert you can not use Disable RAPI command $FD. You can use it but if you put it to disable it is going back to charge if has FV available so you can not stop the charge while FV is available
> 
> Best regards

The same thing happens to me! 

In that case, sometimes openevse even continuously switches between Charging and No Charging.

glynhudson, 2020-09-30:
Yes, EcoMode will always charge when solar PV is avaialble. EcoMode can be disabled by MQTT:

To enable EcoMode `{base-topic}/divertmode/set 2`

To disable EcoMode (Normal Charging, which will not restart the charge) `{base-topic}/divertmode/set 1`

closed 2020-09-30 by glynhudson

ayasystems, 2020-09-30:
Hi, thanks for your answer

It is a new behavor in 2.9.1 prior to this version it was not working in this way.

I think disable must be work without see other mode. If openevse is disable nothing more should be looked ignoring pv-divert mode.