OpenEVSE/ESP8266_WiFi_v2.x #266: Current per phase

snitkjaer, 2021-01-19:
Is it possible to show the current per phase?

It seems to me that not all cars charge using all 3 phases through out the charge.
As far as I can tell the Zoe only uses one phase for the final part of the charge which means the energy calculation is wrong.

glynhudson, 2021-01-19:
Sorry this is not possible with current OpenEVSE hardware. 

I would have assumed an EV charger would draw similar current on each phase, interesting you mention the Zoe doesn't. I would be interested to see some monitoring on this.

closed 2021-01-19 by glynhudson

snitkjaer, 2021-01-19:
I have a attached a screenshoot from my Home Assistant graph where you can see that the home consumption goes negative as the calculated EV charging power (voltage * current * 3) is higher than to measured home consumption (Kostal energy meter)
[image]

When I charge my Tesla this graph looks just fine. I guess the current measurement is not that precise either.