OpenEVSE/ESP8266_WiFi_v2.x #268: Eco to overnight boost mode

shakisha, 2021-03-27:
Hello,

i have got a solar installation so during the day I want to charge using the solar power instead of grid electricity. 
I have seen some firmware improvements in 2.9.0 (6e05310) , which if solar power is not sufficient the charger will go automatically into normal mode during the peak off times (night for example). 

Ok, but what will happen if at the end of this night period, the car is still not fully charged? Will the charger go again to eco mode or will it stay in normal?

paulbernardr, 2021-03-27:
Hi Shakisha
I have written a python script that runs on a pi zero with a GFX-HAT display. I have two EV's and it controls the charging to fit under the PV generation curve,. It also allows a night boost facility mode so you can tell the controller you want a night charge to kick in eg. for Octopus Go from 12.30am to 4.30am for those dull winter months of the year. However we charge 2 EV's on shunshine only for about 9 months of the year.
The pi zero is in our hall and is used by my wife who is in no way techy! She's always out in her Zoe as it's 'free' travel.

jeremypoulter, 2021-03-27:
The new v4 firmware for the ESP32 is able to do this: https://github.com/OpenEVSE/ESP32_WiFi_V4.x