OpenEVSE/ESP8266_WiFi_v2.x #277: RJ45 for wired ethernet connection.

doanerock, 2022-05-04:
There are many reason why you would not want to use wi-fi.  The access point can be  too far away, there can be too many IOT devices which is causing causing interference.   You would rather not put the EVSE on wi-fi for security reasons.

glynhudson, 2022-05-05:
We have a wired Ethernet option: https://shop.openenergymonitor.com/openevse-etherent-gateway-esp32/

closed 2022-05-05 by glynhudson

doanerock, 2022-05-16:
Sorry I did not realize there where stores selling OpenEVSE besides
https://store.openevse.com/collections/all-products

Is the set up any afferent then the only wi-fi version?

glynhudson, 2022-05-16:
>Is the set up any afferent then the only wi-fi version?

There's obviously no WiFi setup require on the Ethernet version, just plug in the Ethernet cable, and it will get assigned an IP address via DHCP. This IP is displayed on the LCD at startup. The web config can then be accessed via the IP address of hostname e.g http://openevse-XXXX where XXXX is the ID number of the unit which is also displayed on LCD at startup. 

The firmware is pre-loaded.